perexg/satip-axe #155: Build 16

perexg, 2019-10-08:
**satip-axe-201910081157-16** : http://s000.tinyupload.com/?file_id=55622426912282482797

- remove very old minisatip and minisatip5
- update minisatip to 1.0.2
  - it's controlled using the original options in /etc/sysconfig/config
<pre>
     MINISATIP="yes"
     MINISATIP_OPTS=""
</pre>
- oscam rev 11546

rkorn86, 2019-10-11:
Could you make a github Release, so that upgrade-fw can be used?

perexg, 2019-10-12:
It's pre-alpha release. Just copy .fw file to /root and run:
<pre>
upgrade-fw /root/*.fw
</pre>

paluseb, 2019-10-24:
I'm trying to update but I'm still 15 after the reboot.

okmikel, 2020-02-09:
Using it for some time now without any issue. Looks very good and stable for a pre-Alpha. 

Looking forward for a alpha or beta :)

NemoN, 2021-09-16:
any other DL source for satip-axe-201910081157-16 ?

okmikel, 2021-09-16:
> Web Bug from
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "name": "View Issue" }, "description": "View this Issue on GitHub",

Try the fork:

https://github.com/Jalle19/satip-axe

He has made new builds with newer minisatip.