perexg/satip-axe #162: please help with direct reception testing

jendik, 2019-12-22:
Hi There and thanks a lot for all the work, not just here-)

I've just got DIGIBIT R1 and wanted to make it woorking with tvheadend... no luck with original firmware, nor with the satip-axe one... I tried almost all combinattions of config parameters in TVH and "OPTS" I was able to put together or googled.

As I suspect wrong LNB configuration with diseqC comobination, I'd like to ask you for any working link/URL for testing any free-to-air channel from Astra23,5 or 19,2 via vlc...

Could you please share any working "URL" to be opened with VLC, just to be sure, DIGIBIT can tune, lock and stream the channel?

btw, could you please help me what is the proper configuration parameter/settings for:
1xQuad LNB with diseqC - that means 4 coax cables to each DigiBIT tuner from satellite parabola, where diseq position1 is Astra23,5 and position2 is Astra19,2 at each cable/tuner?

Thanks so much for your help
Kind regards
J.

9000h, 2019-12-22:
http://www.satip.info/resources?tid=23

jendik, 2019-12-23:
thanks so much for super fast answer... I tried this(with my proper IP):
`rtsp://sat.ip/?src=1&freq=11347&pol=v&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6600,6610,6620,6630`
 and VLC connects, but no video, no audio... minisatip-axe web status says SNR 0%, so I guess there is no lock...?
could youo tell me please, how digibit box with satip-axe recognize which disecq signal should be sentt to tune proper transponder at proper LNB please?

thanks a lot
j.