perexg/satip-axe #165: Cannot get Unicable set-up to work properly

synegic, 2019-12-30:
Hello there,

First off, thanks for this great piece of software, it really helps unlocking full potential of the SATIP server hardware. Hope you can help me with the following which I can't seem to find the right answer to, even after days of trying out almost all possible configuration and firmware/build combinations.

What I am trying to accomplish is having 3 separate Unicable II LNB's connected to physical inputs 0-2, while still being able to use input 3 for a 'normal' LNB. With the newer fw builds, it seems the -9 and -X options are missing completely, so I've been unable to actually use more than one Unicable input so far with these builds. If attempting to setup input mapping by using the -A free input mode, it doesn't seem to be possible to combine this with the Unicable inputs. It always seems to take LNB1 (input 0) when these adapters are set to a Unicable slot and channel with the -u option, regardless of the src/SATIP server value in TVheadend. Changing the network group number also does nothing for the Unicable inputs.

Ex; the following did not work and only tuned to LNB1 for Unicable on adapters 1-3;
`-q *:1500-54-15-15-15-0 -A 0:0:0,1:1:0,2:2:0,3:3:0 -W 1 -u 0:1-1420,1:0-1210,2:2-1680 -b 4042752 -B 10000`

I downgraded to build 14, and there was indeed a -9 option available on minisatip5. However, when using this I am only able to use one tuner per input as it seems (input 0 -> adapter 0, input 1 -> adapter 2, etc). Also, because of using -9, it seems all inputs are set to Unicable and the normal LNB stops working as it simply assigns remaining adapters to the last given Unicable group.

This seemed to semi-work, but only with one adapter per input;
`minisatip5 -q *:1500-54-15-15-15-0 -U 0,1,3 -b 4042752 -B 10000 -u 0:1-1420,1:0-1210,3:2-1680 -v adapter -f`

Things i've tried so far; all firmware releases from latest build 16 back until build 14 with multiple versions of minisatip and also multiple configurations - also in TVHeadend. Nothing seems to bring the solution I'm hoping for. Ideally, I'd like to be able to;

- Use physical inputs 0, 1, 2 for separate Unicable LNB's
- Use adapter 1, 2, 3 for any of the Unicable inputs
- Use physical input 3 for a normal LNB
- Use adapter 4 for the normal LNB on input 3

Would this be possible at all, and if so, how would I go about configuring this properly?

Thanks in advance!

PaperLawyer, 2020-03-07:
Hi, I want to do exactly the same thing as you - was this issue resolved? I do not wish to upgrade my set up to a twin unicable lnb arrangement if minisatip cannot accommodate it.

Jalle19, 2020-06-23:
I figured it out. The `-9` shorthand option doesn't actually work in minisatip8, you have to use `--dmx-source`. Additionally, the format is not what the README says. In order to create the configuration specified in the README:

```
box input 0 ---+--- wire 1 (unicable group 1)
box input 1 ---/

box input 2 ---+--- wire 2 (unicable group 2)
box input 3 ---/
```

You have to use `--dmx-source 0-1:0,2-3:2`. minisatip will then print this when starting:

```
[23/06 12:43:40.269 main]: Setting frontend 0 for adapter 0
[23/06 12:43:40.270 main]: Setting frontend 0 for adapter 1
[23/06 12:43:40.270 main]: Setting frontend 2 for adapter 2
[23/06 12:43:40.270 main]: Setting frontend 2 for adapter 3
```

Jalle19, 2020-06-23:
@perexg it seems the `-9` switch doesn't actually work (minisatip8 says "unknown option"), the documentation should probably be updated or the software should be changed to reflect this.

synegic, 2020-10-31:
@Jalle19 thanks for your reply. I see you are using a newer version of minisatip (https://github.com/perexg/satip-axe/issues/167) and are using the -U parameter patch to limit adapters to certain sources. Would you be able to give a new example configuration based on that?

Jalle19, 2020-11-14:
@synegic what exactly do you want an example of? This is my personal configuration at the moment:

> minisatip -f -g -j 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -Z *:0 --dmx-source 0-2:0,3:3 -U 0-2:3

- `-j` tells each tuner to use a different Unicable slot and frequency
- `-Z *:0` tells it to keep tuners alive (the LNB takes some time to activate itself otherwise)
- `--dmx-source` tells tuners 0-2 to use coaxial input from the LNB1 port, and tuner 3 to use the input from the LNB4 port
- `-U` tells it that SAT>IP source 1 is available on tuners 0-2, while SAT>IP source 2 is only available on tuner 3

roisiano, 2020-12-14:
Sorry if this is not the "thread" where I can post a question, but...

If I want to:
- Use an Unicable LNB from Astra 19.2E connected to physical tunner 1 for "virtual tunners" 0,1,2
- Use a diseq Astra 19.2E / Hispasat 30W connected to physical tunner 4 for "virtual tunner" 3
 
In this way, I could use 3 "virtual" tunners (0,1,2) for Astra 19.2E (any band H/V), and the last "virtual" tunner (3) for Astra or Hispasat.

Is this possible? How should I configure minisatip options for Digibit R1?

Kind regards

Jalle19, 2020-12-15:
> Is this possible? How should I configure minisatip options for Digibit R1?

Yes, this should be possible. I don't have any DiseqC sources so I can't say for certain how you should configure minisatip, but something like this:

- `-j 0:0-1210,1:1-1420,2:2-1680` tells it to configure three Unicable slots with respective frequencies
- `--dmx-source 0-2:0,3:3` tells it to use the LNB1 physical input for adapters 0, 1 and 2, and the LNB4 physical input for adapter 3. This means you'll have three adapters using the Unicable input (each with separate slot and frequency, as specified by `-j`) and one adapter using the "normal" input.
- `-U 0-3:2` tells it that SAT>IP source (`src`) 1 (19.2E) is available on all adapters and source 2 is available only on adapter 3.

Finally, in tvheadend (I can't speak for other clients) you would configure tuners 1-3 to have Position 1 enabled and assigned to Astra 19.2E network, while tuner 4 would have Position 1 and 2 enabled, where 1 is Astra 19.2E and 2 is Hispasat 30W. Positions in tvheadend map to sources in SAT>IP.

roisiano, 2020-12-15:
>     * `-U 0-3:2` tells it that SAT>IP source (`src`) 1 (19.2E) is available on all adapters and source 2 is available only on adapter 3.

> Yes, this should be possible. I don't have any DiseqC sources so I can't say for certain how you should configure minisatip, but something like this:
>
>     * `-j 0:0-1210,1:1-1420,2:2-1680` tells it [...]

Thank you very much for the reply!

I think this is a mistake:

"-U 0-3:**2** tells it that SAT>IP source (src) 1 (19.2E) is available on all adapters and source 2 is available only on adapter 3."

In that case, it would be:
-U 0-3:**3**
Isn't it? 

So, I guess that I should change (use) this in:
/etc/sysconfig/config

MINISATIP5="yes"
MINISATIP5_OPTS="-j 0:0-1210,1:1-1420,2:2-1680 --dmx-source 0-2:0,3:3 -U 0-3:3"

Is that right or should I use other different config (other minisatip version, other parameters...)?

Following your explanation, I guess that if I wanted to use:
1) An Unicable LNB from Astra 19.2E connected to physical tunner 1 for "virtual tunners" 0,1
2) A diseq Astra (legacy) 19.2E / Hispasat 30W connected to physical tunner 3 for "virtual tunner" 2
3) A second diseq Astra (legacy) 19.2E / Hotbird 13E connected to physical tunner 4 for "virtual tunner" 3

then, in that case, the config should be:
/etc/sysconfig/config

MINISATIP5="yes"
MINISATIP5_OPTS="-j 0:0-1210,1:1-1420 --dmx-source 0-1:0,2:2,3:3 -U 0-3:2:3"

Is that right too?

Thanks again and kind regards

Jalle19, 2020-12-15:
> "-U 0-3:**2** tells it that SAT>IP source (src) 1 (19.2E) is available on all adapters and source 2 is available only on adapter 3."
> 
> In that case, it would be:
> -U 0-3:**3**
> Isn't it?

Yes, you're right. I'll update my post.

> MINISATIP5="yes"
> MINISATIP5_OPTS="-j 0:0-1210,1:1-1420,2:2-1680 --dmx-source 0-2:0,3:3 -U 0-3:3"
> 
> Is that right or should I use other different config (other minisatip version, other parameters...)?

The options look correct, however:

1. AFAIK the `-U` option is only available in newer minisatip versions, so you'll need to use my forked build from here: https://github.com/Jalle19/satip-axe/releases/tag/build16
2. Use `MINISATIP` instead of `MINISATIP5` (i.e. no numerical suffix).

roisiano, 2020-12-15:
I'm using now:
https://github.com/perexg/satip-axe/releases

in USB method (NOT in flash).

Build 15 - fix 1 
released this on 21 Oct 2018

a) Could I use that version for this purpose?

b) In case that it's not possible, how to update (if possible; always in USB, not in flash)?
 

c) In case I use this:
https://github.com/Jalle19/satip-axe/releases/tag/build16

Should I be worried about this?

"Remove tvheadend and Python support in an attempt to make this easier to maintain"

I use tvheadend to "manage" Digibit R1.

"Disable SAT>IP client support in"

Jalle19, 2020-12-16:
> a) Could I use that version for this purpose?

I think the `-U` option is not available in that build unfortunately.

> b) In case that it's not possible, how to update (if possible; always in USB, not in flash)?

You just put new files on your USB drive and reboot the device.

> Should I be worried about this?
> 
> "Remove tvheadend and Python support in an attempt to make this easier to maintain"
> 
> I use tvheadend to "manage" Digibit R1.
> 
> "Disable SAT>IP client support in"

No this won't affect you in any way.

roisiano, 2020-12-16:
I have been reading this:
https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md#free-inputs

about option for "free inputs" (-A), and it seems, perhaps, that it would be even better (... or not, I don't know), and even more "cable free", but I don't understand how could I use in this case. If it would be possible to use it, perhaps again, it could be possible to use "less cable" and without diseqCs.
 
Could be possible use -A options to:
- Connect only a cable (Unicable II from Astra) to LNB1 of Digibit R1
- Connect only a cable (universal LNB from Hispasat) to LNB3 of Digibit R1
- Connect only a cable (universal LNB from Hotbird) to LNB4 of Digibit R1

and using "all the tunners with all the positions" (and in Astra, with all bands and H/V, because of the Unicable II).

Reading about free-inputs, perhaps it's possible, but in the example there is no Unicable, and some diseqCs, so... I'm not sure.

Jalle19, 2020-12-16:
I had a look at "free inputs" too at some point but I also came to the conclusion that it's incompatible with Unicable. I might be wrong though. I haven't been able to have two physical Unicable inputs shared freely among all four tuners either - it seems the only way to make it work is to "dedicate" certain tuners to each Unicable input. If someone has a solution to this, please speak up!

roisiano, 2020-12-16:
If unicable is not compatible with -A, I have a "crazy" idea.

I have bought an Unicable LNB for Astra 19.2 (GigaBlue Ultra Unicable SCR-LNB): https://www.amazon.es/GigaBlue-Ultra-Unicable-SCR-LNB-multialimentaci%C3%B3n/dp/B084662RQ1

This LNB has two "legacy outputs". 

Could I use the 2 "legacy outputs" (cables A and B: A to LNB1; B to LNB2 of Digibit R1), an universal LNB cable (C) from Hispasat to LNB3, and an universal LNB cable (D) from Hotbird?

Could it be possible using some "-A" config to:

- Watch channels from different Astra transponders (H and V): up to 4 tunners
- Watch up to 4 different Hispasat transponders (but all of the same band and polarity) 
- Watch up to 4 different Hotbird transponders (but all of the same band and polarity) 

Of course, the number of total tunners available would be 4, but, for example, I could be watching 10500V (cable A from LNB1) and 12500H from Astra (cable B from LNB2), 11500V from Hispasat (cable C from LNB3) and 12000H from Hotbird (cable D from LNB4). But, if you're not using any Hotbird or Hispasat transponders, you could be watching 10500V + 10600V (cable A from LNB1) / + 12500H + 12600H (cable from LNB2) from Astra , etc.

Is that possible or only "crazy"?
 
It would be a curious case; I would have bought an Unicable LNB to use the Legacy outputs.

Jalle19, 2020-12-16:
I honestly don't know. Let me know if you figure it out!

roisiano, 2020-12-18:
Can I use any option when I connect via ssh with Digibit R1 with alternative firmware to see "what is happening" with minisatip (errors in configuration, errors in hardware, something like that)?

Jalle19, 2020-12-18:
Yes, you can run minisatip in the foreground so you can see what it's doing.

roisiano, 2020-12-18:
> Yes, you can run minisatip in the foreground so you can see what it's doing.

Plase, could you tell me how to (commands), or, perhaps even better, how to generate an "activity log" and watch it?

Kind regards

Jalle19, 2020-12-21:
Here's an example: https://minisatip.org/forum/viewtopic.php?p=2443&sid=2e97c834118f08c43f28ff0b6fc8c709#p2443 (remember to use `-f` when running minisatip)