perexg/satip-axe #167: [question] new stable version planed

9000h, 2020-03-11:
Hi @perexg,
do you plan a new stable version with the current minisatip git?
CU
9000h

Jalle19, 2020-07-07:
I built the latest minisatip for myself, will see how it runs and report back. I can post binaries if it seems to work okay.

9000h, 2020-07-07:
Hi,
don't worry, if needed I can do it by myself.
CU

flotux2, 2020-09-13:
Could someone please provide the firmware with latest minisatip included?

Jalle19, 2020-09-16:
@flotux2 try this one: https://mega.nz/file/fA9kxYRD#FKE_Hj9pzKHR8wY4SmReolA2GOFD5UMoRcLdZfMhU3Q

It's built on top of this patch: https://github.com/catalinii/minisatip/pull/764

flotux2, 2020-09-16:
@Jalle19 thanks alot!  Do you notice any difference (hopefully positive) between this new version and the last official one?

flotux2, 2020-09-16:
With version satip-axe-202009161311-15 minisatip7 cannot be started. I can see countless entries like the following when executing logread:

`Jan  1 12:01:20 (none) local0.notice root: minisatip7 exited 127, restarting`

dmesg doesn't show any strange message.

With version satip-axe-201810211549-15.fw I never had these problems.

Any idea why that happens?

Jalle19, 2020-09-16:
My build only has `minisatip`, there is no `minisatip7` or `minisatip8`. Make sure you update your /etc/sysconfig files too to reflect this.

> Do you notice any difference (hopefully positive) between this new version and the last official one?

I haven't noticed any negative issues at least, and the web interface has some more details. minisatip is regularly maintained so it's always a good thing to be able to use the latest version.

Jalle19, 2020-09-16:
Actually, I forgot to mention that I successfully use https://github.com/catalinii/minisatip/pull/736 now, it wasn't possible with the previous build.

posixx, 2020-09-30:
@Jalle19 i installed you build, made sure only minisatip was enabled in config file:

MINISATIP="yes"
MINISATIP_OPTS=""

It appears minisatip is running:

Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.26.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 12:00:01 (none) daemon.info init: starting pid 718, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 719, tty '': '/etc/init.d/ntpd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 720, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 721, tty '': '/etc/init.d/satip'
Jan  1 12:00:01 (none) daemon.info init: starting pid 722, tty '': '/etc/init.d/telnetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 723, tty '': '/etc/init.d/dropbear'
Jan  1 12:00:01 (none) daemon.info init: starting pid 724, tty '': '/etc/init.d/inetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 725, tty '': '/etc/init.d/crond'
Jan  1 12:00:01 (none) daemon.info init: starting pid 727, tty '': '/etc/init.d/oscamd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 728, tty '': '/etc/init.d/customd'
Jan  1 12:00:01 (none) daemon.err udhcpc[718]: started, v1.26.2
Jan  1 12:00:01 (none) local0.notice root: DHCP deconfig
Jan  1 12:00:01 (none) authpriv.warn dropbear[723]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan  1 12:00:01 (none) authpriv.warn dropbear[723]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan  1 12:00:01 (none) cron.info crond[725]: crond (busybox 1.26.2) started, log level 8
Jan  1 12:00:01 (none) authpriv.info dropbear[723]: Not backgrounding
Jan  1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan  1 12:00:02 (none) daemon.warn mountd[720]: Kernel does not have pseudo root support.
Jan  1 12:00:02 (none) daemon.warn mountd[720]: NFS v4 mounts will be disabled unless fsid=0
Jan  1 12:00:02 (none) daemon.warn mountd[720]: is specfied in /etc/exports file.
Jan  1 12:00:02 (none) daemon.notice mountd[720]: Version 1.3.3 starting
Jan  1 12:00:02 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:05 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:08 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:11 (none) local0.notice root: DHCP leasefail
Jan  1 12:00:31 (none) daemon.err udhcpc[718]: sending discover
Jan  1 12:00:32 (none) daemon.err udhcpc[718]: sending select for 192.168.50.162
Jan  1 12:00:32 (none) daemon.err udhcpc[718]: lease of 192.168.50.162 obtained, lease time 7200
Jan  1 12:00:32 (none) local0.notice root: DHCP bound ('192.168.50.162' '' 'netmask 255.255.255.0' | '192.168.50.254' '192.168.50.250') replacing ''
Jan  1 12:00:32 (none) daemon.notice minisatip[984]: minisatip version 1.0.3-abba905, compiled in Sep 16 2020 12:49:14, with s2api version: 0501
Jan  1 12:00:32 (none) daemon.notice minisatip[984]: Initializing with 4 devices
Sep 30 06:55:41 (none) cron.err crond[725]: time disparity of 3022255 minutes detected
Sep 30 06:56:33 (none) auth.info login[987]: root login on 'pts/0'
Sep 30 07:55:13 (none) daemon.err udhcpc[718]: sending renew
Sep 30 07:55:13 (none) daemon.err udhcpc[718]: lease of 192.168.50.162 obtained, lease time 7200
Sep 30 07:55:13 (none) local0.notice root: DHCP renew ('192.168.50.162' '' 'netmask 255.255.255.0' | '192.168.50.254' '192.168.50.250') replacing '192.168.50.162'
Sep 30 07:57:50 (none) auth.info login[1026]: root login on 'pts/0'

But my TVHeadend is not discovering the tuners. It worked when i ran the stock firmware of my Digibit R1. 

An i doing something wrong here?

posixx, 2020-09-30:
hmm after restarting TVHeadend the minisatip adapters are lost again.. something is wrong here..

Jalle19, 2020-10-01:
Yes, this has nothing to do with this new firmware version.

flotux2, 2020-10-02:
> My build only has `minisatip`, there is no `minisatip7` or `minisatip8`. Make sure you update your /etc/sysconfig files too to reflect this.

Ok, I will change my sysconfig accordingly and will test again.

Btw: do you have any plans to fork maybe this repository and create new builds with the latest minisatip version on a regular basis? That would be really awesome. perexg seems to have given up further development.

perexg, 2020-10-02:
> Ok, I will change my sysconfig accordingly and will test again.
>
> Btw: do you have any plans to fork maybe this repository and create new builds with the latest minisatip version on a regular basis? [...]

Not really, but the current minisatip covers all my needs and the new versions are usually slower. I'll try to do some updates this month.

flotux2, 2020-10-04:
I changed my settings in /etc/sysconfig/config to:

MINISATIP="yes"                                               
MINISATIP_OPTS="" 

I cannot view any channel with DVBViewer or an Android SAP>IP client.

dmesg shows the following entries:

```
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
Unaligned userspace access in "minisatip" pid=1028 pc=0x004251b8 ins=0x0826
Unaligned userspace access in "minisatip" pid=1028 pc=0x004251b8 ins=0x0826
Unaligned userspace access in "minisatip" pid=1028 pc=0x004251b8 ins=0x0826
Unaligned userspace access in "minisatip" pid=1028 pc=0x004251c8 ins=0x1811
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 0
[dms-ts] stream stop ok
[dms-ts] stream stop ok
```

Jalle19, 2020-10-05:
Please ask for generic help with minisatip somewhere else, you are most likely missing relevant options since your `MINISATIP_OPTS` is empty.

flotux2, 2020-10-05:
> 
> 
> Please ask for generic help with minisatip somewhere else, you are most likely missing relevant options since your `MINISATIP_OPTS` is empty.

A valid statement I would say. With version 0.7 of your latest build from 2018 I also had no additional options and it worked without problemlos. So I will use that version.

Jalle19, 2020-10-05:
Okay. I guess I'll have to try the latest build myself.

Jalle19, 2020-11-14:
I upgraded both my boxes to my latest build and it works fine. I'll try to get an actual release made with change log and everything, uploaded to GitHub.

estimadarocha, 2020-11-14:
waiting for it!

Jalle19, 2020-11-14:
https://github.com/Jalle19/satip-axe/releases/tag/build16 here it is!

flotux2, 2020-11-14:
@Jalle19: with you latest build I get on my Grundig GSS.BOX still these error messages and no channel works. :-(

`Unaligned userspace access in "minisatip" pid=1028 pc=0x004251b8 ins=0x0826`

Jalle19, 2020-11-14:
It didn't work before either?

Jalle19, 2020-11-14:
No idea why you're getting "Unaligned userspace access" but it's surely a minisatip bug, not a bug in these builds in general.

flotux2, 2020-11-14:
It only worked with the minisatip 0.7 builds of perexg. Your builds with the 1.x versions of minisatip never worked on my Grund hardware.

Jalle19, 2020-11-14:
Okay. Sorry to hear that, unfortunately I don't have the time nor skills to debug these kinds of issues. My main motivation behind getting new builds out were because I need them myself.

Jalle19, 2020-11-14:
Since minisatip doesn't seem to crash for you (judging by your earlier dmesg log snippet) the problem is likely that the default options have changed, and since you're not specifying any options (`MINISATIP_OPTS=""`) your at the mercy of whatever the defaults are.

I suggest opening an issue here instead: https://github.com/catalinii/minisatip/issues

PsikoBlock, 2020-11-16:
Same problem as @flotux2 with `MINISATIP_OPTS=-Q`. 
```
@@@@@@@@@@@@ fe_open 0
Unaligned userspace access in "minisatip" pid=970 pc=0x0041f2ec ins=0x0826
```
(repeated 9 more times)
```
Unaligned userspace access in "minisatip" pid=970 pc=0x0041f2f6 ins=0x2812
Unaligned userspace access in "minisatip" pid=970 pc=0x0041f2fc ins=0x1811
```
(everything repeated for fe_open 1, 2 and 3)

On the older build (201910081156-16) I'm getting (again 4 times)
```
Unaligned userspace access in "minisatip" pid=971 pc=0x00429f52 ins=0x2872
Unaligned userspace access in "minisatip" pid=971 pc=0x00429f56 ins=0x1811
```
however it works fine.

My guess is it has to do with opening the adapters, as there are 4 of them. Sadly I don't have that much time for debugging either, but am definitely willing to try out new fixes :)

Jalle19, 2020-11-16:
So everything is working despite those log messages?

PsikoBlock, 2020-11-16:
On the old build, yes. On your new build, streams are not working.

CJanne, 2020-11-16:
> https://github.com/Jalle19/satip-axe/releases/tag/build16 here it is!

DVBViewer won't work!
I use a simple Unicable config. 
MINISATIP_OPTS="-j 0:0-1210,1:1-1420,2:2-1680,3:3-2040"

HTTP stream works (Webinterface)
RTSP stream don't work (webinterface)

Jalle19, 2020-11-17:
I just checked and I'm also getting "unaligned userspace access" in dmesg, but it doesn't affect functionality. `logread | grep minisatip` shows this:

```
# logread | grep minisatip
Jan  1 12:00:07 (none) user.notice minisatip: Setting jess adapter 0 slot 4 freq 984 pin -1
Jan  1 12:00:07 (none) user.notice minisatip: Setting jess adapter 1 slot 5 freq 1020 pin -1
Jan  1 12:00:07 (none) user.notice minisatip: Setting jess adapter 2 slot 6 freq 1056 pin -1
Jan  1 12:00:07 (none) user.notice minisatip: Setting jess adapter 3 slot 7 freq 1092 pin -1
Jan  1 12:00:07 (none) user.notice minisatip: Set default timeout to 0
Jan  1 12:00:07 (none) user.notice minisatip: Setting frontend 0 for adapter 0
Jan  1 12:00:07 (none) user.notice minisatip: Setting frontend 0 for adapter 1
Jan  1 12:00:07 (none) user.notice minisatip: Setting frontend 0 for adapter 2
Jan  1 12:00:07 (none) user.notice minisatip: Setting frontend 3 for adapter 3
Jan  1 12:00:07 (none) user.notice minisatip: Calculating adapter sources: [0-2:3] 
Jan  1 12:00:07 (none) user.notice minisatip: Adapter correct sources format parameter
Jan  1 12:00:07 (none) daemon.notice minisatip[859]: minisatip version 1.0.4-eef7333, compiled in Nov 13 2020 13:11:13, with s2api version: 0501
Jan  1 12:00:08 (none) daemon.notice minisatip[859]: Initializing with 4 devices
```

If minisatip starts correctly (i.e. it doesn't repeatedly crash) then please open issues at https://github.com/catalinii/minisatip/issues instead. Also verify that you've updated the configuraton (`/etc/sysconfig/config`) to use `MINISATIP_OPTS`, not `MINISATIP7_OPTS` or any other numbered options variable. I'll update the release notes to reflect this change.

ecdlguy, 2021-01-12:
Hi, just for clarification and because I'm a little confused now - 
From your Release Notes of [build 16](https://github.com/Jalle19/satip-axe/releases/tag/build16)
> - Remove tvheadend and Python support in an attempt to make this easier to maintain
> - Disable SAT>IP client support in

Does this mean RTSP is disabled? What do you mean "Disable SAT>IP client support in"? 
Thanks and cheers,
Thorsten

Jalle19, 2021-01-14:
> Does this mean RTSP is disabled? What do you mean "Disable SAT>IP client support in"?

No, it just means that you can't use minisatip in client mode (basically chaining multiple minisatip instances).

If someone actually uses this on these devices I can revert the change.

ecdlguy, 2021-01-15:
Ok I've upgraded to your build 16 and like the other guys @CJanne, @PsikoBlock and @flotux2 RTSP is not working anymore. 

Hardware ist a Digibit R1, Client is VDR with vdr-satip/2.4.1-GIT-v219-3-g39b8f80 plugin.
The minisatip webinterface shows that the tuner is used by the client, however the webIF stats say:
`BW 2KB/s | Total BW 3MiB | ns/read 72 | reads 209 | writes 9 | **failed writes 208** | tt 15ms`

PS: Regarding your previous post, I still don't know what you mean by "Remove tvheadend [...] support"

cheers, Thorsten

Jalle19, 2021-01-15:
> Ok I've upgraded to your build 16 and like the other guys @CJanne, @PsikoBlock and @flotux2 RTSP is not working anymore.

It could be related to this: https://github.com/Jalle19/satip-axe/issues/3. Can you check your `logread` output if you have those messages there?

ecdlguy, 2021-01-15:
I just went back to 201910081156-16 to get the TVs running again, I'm going to flash your version again in the next few day and report back. Sorry many scheduled recordings ahead I do for a friend of mine.

estimadarocha, 2021-01-31:
@Jalle19 any news releases planned with the last fixes?

Jalle19, 2021-01-31:
> @Jalle19 any news releases planned with the last fixes?

Yes, I'm working on it.

Jalle19, 2021-01-31:
@estimadarocha @ecdlguy @CJanne @PsikoBlock @flotux2 https://github.com/Jalle19/satip-axe/releases/tag/untagged-386e81cdb9e2ba1ba9ca can you test this one?

PsikoBlock, 2021-01-31:
> https://github.com/Jalle19/satip-axe/releases/tag/untagged-386e81cdb9e2ba1ba9ca

Thanks for your continued work on this! I'd love to test the build, but the link 404s for me.

Jalle19, 2021-01-31:
> > https://github.com/Jalle19/satip-axe/releases/tag/untagged-386e81cdb9e2ba1ba9ca
> 
> Thanks for your continued work on this! I'd love to test the build, but the link 404s for me.

Sorry about that, link should work now!

CJanne, 2021-01-31:
> Update minisatip to 191fe62a7a5aaada03ef274511b24238c210693c, should fix sending of UDP packets (#3)

RTSP streaming is working now!
Thx!

jacotec, 2021-02-02:
@Jalle19 Thank you that you are taking care of this great firmware! :)
What means "Remove tvheadend and Python support in an attempt to make this easier to maintain" ... what was the special treatment for tvheadend (as I'm using TVH as TV server with my Digibit R1)? Any drawbacks expected with your version?

And ... is OSCAM still included and what version?

estimadarocha, 2021-02-02:
@Jalle19 

minisatip -f -g -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 4042752 -B 10000
RECEIVED SIGNAL 11 - SP=41502E IP=2961DA14
Obtained 1 stack frames.
sh: addr2line: not found
0x415fe2 :

Jalle19, 2021-02-03:
> What means "Remove tvheadend and Python support in an attempt to make this easier to maintain" ... what was the special treatment for tvheadend (as I'm using TVH as TV server with my Digibit R1)? Any drawbacks expected with your version?

Sorry for the confusion, it really should read "Removed Python support and tvheadend". The only thing related to tvheadend that's been removed is an obsolete build for running tvheadend itself on the device.

> And ... is OSCAM still included and what version?

Same as before, https://github.com/Jalle19/satip-axe/blob/master/Makefile#L68

@estimadarocha please open an issue at https://github.com/catalinii/minisatip instead.

ecdlguy, 2021-02-19:
> @estimadarocha @ecdlguy @CJanne @PsikoBlock @flotux2 https://github.com/Jalle19/satip-axe/releases/tag/untagged-386e81cdb9e2ba1ba9ca can you test this one?

Sorry it took so long, but yes, RTSP is working!
Thanks a lot!

estimadarocha, 2021-02-19:
For me also fine

bjoerns1983, 2023-01-26:
@perexg seems that you dont have the time to support the project at the moment.
Wouldnt it be a good idea to link to the fork of @Jalle19 in the readme because he is actively working on the FW?