perexg/satip-axe #172: Unicable Configuration SCR1-SCR24 - not all frequencies are working

satip, 2020-11-05:
Hi,

i changed my Unicable-LNB from 4x SCR to a new one 24x SCR (https://www.amazon.de/dp/B079TRCDZN/ref=cm_sw_r_oth_api_i_KHWOFbY732FY6).

The old LNB was working with:
MINISATIP8_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"

Only some frequencies (cannot say exactly which) are working with the new LNB:
https://pasteboard.co/JyU0TN9.jpg
MINISATIP8_OPTS="-u 0:1-975,1:0-1025,2:2-1075,3:3-1125"

It this a Hardware- or a Software-limit?
Using Version 201810211549-15

thx

catalinii, 2020-11-28:
The implementation is the same regardless of the number of valid inputs. Are some of the frequencies that were working before not working now ?

satip, 2020-12-11:
as you can see from the picture, the new LNB has not the same frequencies than the old one. 
old: 1420 1210 1680 2040
new: xx25 xx75
I have sent back the new LNB but can request it again, if needed for further investigations

closed 2020-12-11 by satip