perexg/satip-axe #174: reboot command boots to original firmware - digibit R1

jendik, 2020-11-16:
Hi There,

thanks so much for the great piece of SW. 

It's probably not an issue...it's more or less question as I probably don't know how to reboot "by using a command via command line" to get to satip firmware again (without not turning off and on). Reboot command brings me to original firmware as it for some reason probably doesn't boot from usb flash disk. The only solution that works for me is to disconnect it from the electricity and connect again to boot satip from usb flash disk.

Could you please advice?

~Thanks so much
Kind regards
Jan

perexg, 2020-11-16:
Unfortunately, due to an issue in the uboot (the boot loader) in this device, the warm boot cannot load the firmware from usb.

Flash the firmware to the device permanently.

jendik, 2020-11-16:
Aha...sure, no problem, thanks a lot for the help, I will give a try flashing.

Have a great day
Thanks so much for all the work!
take care
j.

jendik, 2020-11-18:
Hi There,

one more question please. Is there a way how to remotely upload the another file to the usb flash drive for flashing internal firmware without having usb drive in hand?

I will restart power after that remotely...

Thanks so much
Kind regards
J.

jendik, 2020-12-06:
I asked within another issue as this one is solved, thanks!!!

closed 2020-12-06 by jendik