perexg/satip-axe #177: remote permanent Flash to the device

jendik, 2020-12-06:
Hi There,

Please, is there a way how to remotely upload the another file to the usb flash drive for flashing internal firmware without having usb drive in hand?

I mean is there a way how to mount USB flash disk (FAT32) inserted in the device and booted with sat-ip to be able to upload another firmware file there for permanent flash after reboot please?

I will restart power after that remotely...

Thanks so much
Kind regards
J.

Jalle19, 2020-12-16:
USB sticks are auto-mounted on `/media/USB_DISK/`. It should be possibly to simply place files there, then issue a reboot. Not sure if `reboot` works or if you need to physically cut the power.

jendik, 2020-12-16:
THANKS so much, now I can ssee it. In my case it's here:
/media/sda
/media/sda1

That's good part. I also had to cut the power which is ok as well and was expected.

The issue is that original FW is loaded after flashing and power cut. 
And it looks like the USB will have to be removed from the slot to boot flashed sat-ip firmware. I'm afraid I recall that the same situation occurred when I flashed another box.

That would unfortunately mean physical persistence by the box, which looks like remote flashing is not possible... ?

thanks so by so
j.

Jalle19, 2020-12-16:
I don't see why it would boot to the original firmware?

jendik, 2020-12-16:
I understand, I might have done something wrong and I will recheck when I get by the box. Thank you, will let you know.

fergycool, 2021-05-05:
I can confirm that a reboot command in a SSH session reboots to the original firmware. You have to power cycle to boot from satip-axe on the USB stick.  I'm afraid I have no idea how boot loaders work on devices such as this (Digibit R1 for me) so I am no help in fixing it.

perexg, 2021-05-05:
Apparently, there's a bug in u-boot which prevents to boot from the USB flash (hot reboot). The USB code in uboot does not work properly in this case (the USB flash is not detected). I recommend to write the fw to the flash.

fergycool, 2021-05-05:
Thank you. I'll eventually flash the internal flash. But booted from the USB device so far your tool is very stable!

Jalle19, 2021-05-06:
If the goal is unattended firmware upgrades then one solution is of course to connect the box to a smart plug, and then issue a cold reboot by toggling the power off and on. Not ideal, but definitely doable.