perexg/satip-axe #181: Config for minisatip in Digibit R1 with two Unicable II a single cable

Javito68, 2021-02-03:
Hello greetings to all, is this configuration possible with minisatip?

stage:
  Sat A: Astra 19.2E with Lnb Unicable II inverted.
  Sat B: Hot bird, with another antenna and another inverto unicable Lnb.
  A single cable with a combiner connected to Digibit R1 at input 0, see: Additional scenarios - SDU receiving 2 satellites (https://www.inverto.tv/what-is-unicable-2).
In digibit R1 input 0-1 for Sat A and 2-3 for Sat B.

How should I configure the minisatip options?
the frequencies (scr) in the Lnbs would have to be different?
Has someone managed to make this configuration?

Now I am using:
  Sat A (Astra 19.2E) with lnb Unicable-Digigit R1 (input 0), with version
  minisatip:
  minisatip version 0.7.16-ax215, compiled with s2api version: 0501, minisatip8:
   MINISATIP8_OPTS = "- diseqc-timing *: 155-54-15-15-15-0 -u 0: 0-1210,1: 1-1420,2: 2-1680,3: 3-2040"
It works perfect.

Thanks !!!

Jalle19, 2021-02-03:
I think you need a Unicable programmer to reprogram the LNBs to use different frequencies. I don't have one so I haven't been able to try such a setup myself.

perexg, 2021-02-03:
https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md#multiple-unicable-groups

Javito68, 2021-02-04:
Thanks for answering,
  perexg I mean a single cable, the option you mention is for several. Regarding the Unicable programmer for the Lnb inverto II we have the program option: senddsq in satip-ax / dist / packages /, developed by akosinov: https: //github.com/akosinov/unicable, I have tested it with the Lnb II inverto 32 user band (Model 5278) and it doesn't work for me,
Has someone managed to program an inverto Lnb with senddsq? (I remember I am using: minisatip version 0.7.16-ax215, compiled with s2api version: 0501, minisatip8)
Is there another more updated senddsq binary?

some help....

Thank you all!!!

Jalle19, 2021-02-04:
AFAIK `senddsq` simply sends commands from a file to the LNB. There's no UI or anything like that for doing the actual programming.

Javito68, 2021-02-04:
Hello Jalle19,
  I explain how I have done it:
1- With the inverto application: Unicable II Programmer you can generate files of the type dsq (hex), bin or txt with the frequencies and Ubs (see fig).
  
[image]

2- mediante: dconv (de akosinov,  config formats converter) genero el fichero dsq :
   ./dconv -i IDLU-32UL40_5a32.dsq -it dsq -o idlu_new.dsq -ot dsq -l 6 
  file dsq: 
[image]

3- Envio el fichero dsq con senddsq:
 ./senddsq -f 0 -i idlu_new.dsq

When sending the sequences to the Lnb, with Digibit R1, through input 0 (-f 0) I observe that led 1 of the digibit does not turn on and I think it would have to turn on since when I tune a channel with minisatip8 it turns on (log: send_unicable fd 11, freq 1156, ufreq 1210, pos = 0, pol = 0, hiband = 0, slot 0, diseqc => e0 10 5c 00 f2).

this is correct?
Any suggestion? 
Could the senddsq file be modified?

Thanks !!!

perexg, 2021-02-04:
The LED status may not reflect the tuner use. It's sw controlled. Minisatip controls LEDs, but senddsq does not . You may analyze the dvbapi ioctls with strace.

If you cannot select the position for LNBs (unicable support  2 sat position, unicable II 64 positions):

The SCR frequencies must be different for both LNBs and you cannot share one tuner with multiple LNBs. So LNB A can be assigned for tuners 0 and 1 and LNB B to tuners 2 and 3 for example. Just configure non-collision unicable frequencies for all LNBs connected to the shared coax wire and use the frequency for given LNB.

If you can select position for LNBs:

You can probably share the SCR frequencies, but you must set LNB A == position 0 and LNB B == position 1 for example. But I don't see the position settings in your screenshot.

Javito68, 2021-02-04:
perexg,
  You may analyze the dvbapi ioctls with trace.
   how can i analyze the trace dvbapi ioctls? with /sbin/minisatip8 -l http,dvb ....?

I explain the real situation:
  -sat A with Dur-line unicable Lnb (4 + 1) Not programmable with fixed SCR frequencies: ch01: 1210Mhz, ch02: 14200, ch03: 1680 cho4: 2040 (is the one I have working)
-sat B with unicable Lnb II invert programmable and programming the SCRs (for example with these frequencies): ch09: 1128, ch10: 1164, ch11: 1256, ch12: 1292
see Figure: 
[image]

the parameters in minisat would be (I think):
/ sbin / minisatip8 -l http, dvb --diseqc-timing *: 155-54-15-15-15-0 -u 0: 0-1210,1: 1-1420,3: 3-1128,3: 3 -1164
I would use two SCRs for each satellite (Digibit R1 only 4 tuners).
In such a way that when you select a channel of sat A (tvheadend first slot) minisatip will select the SCR: 0 -1210 and it would only answer the Lnb A since the Lnb B does not have that frequency assigned and idem for the other SCRs.

I think that is how I would have to work, but I was still wrong in this approach.
Anyway the story is in getting to program the inverto Lnb with senddsq.

I will try to try analyze the dvbapi ioctls with trace.

Thanks!!!

Javito68, 2021-02-10:
Hi there,
  some know some unicable Lnb that has fixed SCR frequencies that are different from:
   ch01: 1210Mhz, ch02: 14200, ch03: 1680 ch04: 2040

?
Thanks !!!

Sam-ghub, 2022-02-19:
@Javito68 did you finally manager to config the Unicable with 1 wire?
I'm facing similar config:
LNB 1: non programable EN50494  SCR: 1-2-3-4
LNB2: Progammable SCR: EN50607  disbabled scr (1-8), 10-14 SAT B
LNB3: Progammable SCR: EN50607  disbabled scr (1-8), 10-14 SAT C
I program the LNB with sendsq the LNB and got it working on enigma2 assigning for the same tuner two UB..

I want to keep the 4 tuners without replacing the EN5094 non programmable LNB

Thanks.

https://minisatip.org/forum/viewtopic.php?f=5&t=811