perexg/satip-axe #182: First Tune always fails - No Signal --> Recordings are all Failing

stahlhelmknacker, 2021-03-28:
Hello, 

Thank you all for your support and your hard work. I have the following Setup:

TVHeadend 4.2.8
Digibit R1 with Minisatip/0.7.16-axe215 with the config MINISATIP8_OPTS="-P 0 -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"
Unicable Setup with a single Coax Cable
Connection between Digibit R1 and Linux Server is with a Powerline Adapter.

When I am starting a Channel in Kodi with the TVHeadend PVR Plugin than the first tune always said No Signal. In the Webinterface of Digibit R1 the STR is about 13% and the SNR is 0%. When I am starting another channel and than go back to the first one than the Channel is working flawlessly.

The main problem is that because of this "bug" I cant record anything because it always fails when trying to tune and than the recording is not starting. 

Do you have any ideas how I can configure the setup that my recordings are working?

Thank you very much!

MusterGit, 2021-03-28:
Hi,

I am having the same issue with my setup (R1, Unicable II as single input for all 4 tuners, TVH).
The most common solution that I could find is to keep all tuners active by using the parameter -Z *:0. With this solution you still have to activate the tuners once after each boot by zapping around. For me this has two main disadvantages: 1. the R1 gets really hot, 2. the power consumption is higher. 
Yesterday, I played around a bit with some of TVH's settings and found a better solution for me. In the adapter configuration I changed the timeout setting of the 'Position #1 (AA)' item of each tuner from 10s to 3s, and in the stream profiles I use ('htsp' for watching and 'pass' for recordings) I disabled 'Switch to another service'.

stahlhelmknacker, 2021-03-28:
> Hi,
> 
> I am having the same issue with my setup (R1, Unicable II as single input for all 4 tuners, TVH).
> The most common solution that I could find (see #79, [catalinii/minisatip#313](https://github.com/catalinii/minisatip/issues/313) and [Jalle19#7](https://github.com/Jalle19/satip-axe/issues/7)) is to keep all tuners active by using the parameter -Z *:0. With this solution you still have to activate the tuners once after each boot by zapping around. For me this has two main disadvantages: 1. the R1 gets really hot, 2. the power consumption is higher.
> Yesterday, I played around a bit with some of TVH's settings and found a better solution for me. In the adapter configuration I changed the timeout setting of the 'Position 1 (AA)' item of each tuner from 10s to 3s, and in the stream profiles I use ('htsp' for watching and 'pass' for recordings) I disabled 'Switch to another service'.

Thank you very much for your quick response. I will check all the posts if there is anything that will fix my problem. 

I am recommending you to actively cool your Device with a Fan. I am using the Arctic F12. I have written with Telestar this week because I wanted to buy another Digibit R1 but they told me that they have discontinued the Device because of missing Semiconductor devices that are not produced anymore. Therefore, we need to handle these devices with care!

stahlhelmknacker, 2021-03-28:
Thanks MusterGit for your input regarding the 'Switch to another service' option. I have disables this option in my "pass" Profile and now the Tuning and the Recording is working flawlessly. The first tune takes pretty long but for me the most important part is that the Recordings are working.

closed 2021-03-28 by stahlhelmknacker

MusterGit, 2021-03-29:
Although we found a TVH workaround for our recording issue I don't think that the issue should be closed. The basic problem is still in the way how the tuners are controlled, so probably with minisatip.

MusterGit, 2021-03-29:
Another issue that is related to this issue is the behaviour of an active tuner in case of OTA EPG grabbing. So when I am recording a programme and the OTA EPG grabber scans the transponder of that programme, the tuner shows the same behaviour like with the first tuning and the recording stops. As you seem to have a similar setup, can you confirm this issue?

perexg, 2021-03-29:
You may try to set various diseqc timeouts and repeats, it should help.

List of internal variables in minisatip (maps to `-d` option and `-q` option):

- commited_no - number of repeats (mux map to the unicable freq)
- after_tone - delay after the power / tone is set on the coax
- before_cmd - delay before the diseqc command
- after_cmd - delay after diseqc command
- after_repeated_cmd - delay after repeated diseqc command

MusterGit, 2021-03-29:
No luck so far with these variables. Maybe you can help me.
I'm using a Selfsat H22dCSS+ flat antenna with my Digibit. The dCSS output of the antenna is directly connected to LNB 1 of the Digibit, so there are no further switches in between.

These are my startup parameters so far: 
`-j 0:0-1210,1:1-1420,2:2-1680,3:3-2040 --dmx-source 0-3:0`

Now for the internal variables. My test cycle is: 
1. `nano /etc/sysconfig/config`
2. change parameter
3. `axe-debug reset`
4. evaluate
5. go to 1

-d:    As there are no further switches used, I guess this option can be default (1:0), right?
-q:    Setting the before_cmd to sth high like 1500 does not help. Also setting after_tone to this value shows no effect. Multiplying each value by 10 does not solve the problem either.

What else could I try?

ecdlguy, 2021-03-29:
I'm using `-q *:150-54-15-15-15-0` and it's working fine.

MusterGit, 2021-03-29:
> 
> 
> It's a bad sequence. Just change the config and kill the minisatip task PID.

I re-did my tests with just killing the minisatip process, but the results are the same. Any idea what to change?

> 
> 
> I'm using `-q *:150-54-15-15-15-0` and it's working fine.

For me it does not work, unfortunately. Do you have the same antenna?

ecdlguy, 2021-03-30:
> Do you have the same antenna?

No. But you could try to increase the first value of the "-q" parameter. Probably your antenna needs more time to "boot".
Don't forget to kill your minisatip process after changing the config for the parameters to take effect. You can also check if minisatip is running with the correct paramerts by looking at the process list or just using "top".

cheers, Thorsten

MusterGit, 2021-03-30:
Hi Thorsten, thanks for your response. I already tried to increase that value like I said in https://github.com/perexg/satip-axe/issues/182#issuecomment-809489600. Although this seems to help a lot of people, it doesn't for me.

ecdlguy, 2021-03-30:
Ok sorry I've no idea what else you could try.

MusterGit, 2021-03-30:
Thanks anyway. Have a nice day.

perexg, 2021-03-30:
Try also `-d *:3-0` parameter. It should repeat the "tune" diseqc command three times for unicable / jess. Eventually, try also `-d *:3-3`.

MusterGit, 2021-03-30:
Setting after_tone to 2500 (-q *:15-54-15-15-15-2500) fixes the "first tuning problem", BUT as soon as I tune a second channel (another tuner), I get CC errors for the first stream. This happens whenever after_tone is >0. Setting before_cmd to sth higher than 500 also leads to CC errors as soon as another channel is tuned.
With `-d *:3-0` after_tone can be set to 2250 instead of 2500. `-d *:3-3` does not make any difference.

perexg, 2021-03-30:
2500ms means 2.5 seconds for the LNB wake up. It's too much. The repeating of the diseqc commands also adds some delays, so it explains why you can reduce the initial power time.

MusterGit, 2021-03-30:
Any idea how to proceed? Why do the CC errors appear?

MusterGit, 2021-03-30:
With `-b 4700000` the CC errors do not occur.

stahlhelmknacker, 2021-03-31:
> Another issue that is related to this issue is the behaviour of an active tuner in case of OTA EPG grabbing. So when I am recording a programme and the OTA EPG grabber scans the transponder of that programme, the tuner shows the same behaviour like with the first tuning and the recording stops. @stahlhelmknacker: As you seem to have a similar setup, can you confirm this issue?

Hello MusterGit, I have tried to record a program and trigger an OTA EPG Update with the "Force Scan" in TVHeadend but it seems that i don't have that problem you are describing. I have a "GigaBlue Ultra Unicable SCR-LNB / 24 SCR - 2 Legacy" LNB. 

If you need anything else for testing or any configurations from my side than just ask.

MusterGit, 2021-04-01:
I bought a power inserter and both problems (First Tuning, OTA EPG Scan) are gone. Thanks for your help and support.

perexg, 2021-04-01:
Thanks for the feedback.

estimadarocha, 2023-06-06:
> -d *:3-0

@MusterGit can you share model?

MusterGit, 2023-06-25:
It's a DUR-line Unicable II Power Inserter.

Just installed an update of satip-axe, only to realize the bug is still present in minisatip.