perexg/satip-axe #185: satip-axe does not get a network connection, but stock firmware does

fergycool, 2021-05-12:
I've been using satip-axe on a Digibit R1 for a few weeks now. Booting from a USB stick. Works fine. 
I've just added a third LNB and SSH'ed to edit the sysconfig/config files. All seemed Ok and I made seemingly minor changes to it.
Except it is now inaccessible from the network.
The stock firmware is fine and get's an IP address.
Satip-axe goes through the usual boot routine. Flashing KIT lights, a final blink then just the RED power LED is lit.
I've toggled DHCP on the stock firmware a few times. Added a static Ip address... Nothing.
I've wiped and reinstalled the two firmware files to the USB stick. No change.
How do I access the internal config files? I'm assuming that the network config files are used by the stock firmware AND this custom one?
Would appreciate a few pointers.
Many thanks!

fergycool, 2021-05-12:
I've been looking at my DNS/DHCP server. I use a docker container of Dnsmasq for the DHCP/DNS services on my home LAN. After going through the logs it's definitely not being contacted at all by the Digibit R1. Very, very strange.

fergycool, 2021-05-13:
I've also just noticed that the device is hot when I boot from the USB drive. When booting from the stock firmware it is cool. I wonder if something is causing it to hang at some point.

fergycool, 2021-05-15:
Closing. I'd discounted a corrupted config file early on as I'd used the firmware file to reset the config file. However, I now realise I was not using the right instructions. The updated instructions for resetting the configuration file say to rename the firmware file WITHOUT the version number. I would be very happy to spend time going through the documentation and updating it?

closed 2021-05-15 by fergycool

Jalle19, 2021-05-16:
> I would be very happy to spend time going through the documentation and updating it?

Please, by all means!