perexg/satip-axe #188: Ports for older Linux STB?

SaulGoodman1337, 2021-12-17:
Hi,
is it possible to adapt the image for slightly older linux STB? 
I have an Xtrend ET10000 here that now serves me as a pure SAT IP server.

I would like to have everything else unnecessary removed from the Enigma2 image and so I became aware of the project here. It would be wonderful if the project here could be ported directly to the box

Jalle19, 2021-12-17:
This project is limited to a specific type of hardware, so it's not really that easy to port to a different box. It wouldn't really make much sense either since you can simply run Enigma2 on your device (something that wasn't an option for the boxes this firmware is for).

It's relatively easy to run Enigma2 in "headless" mode and have it start minisatip on startup, there are some useful posts on https://minisatip.org/forum/index.php

SaulGoodman1337, 2021-12-17:
> This project is limited to a specific type of hardware, so it's not really that easy to port to a different box. It wouldn't really make much sense either since you can simply run Enigma2 on your [...]

That is exactly what I am currently doing.
However, the boot time bothers me a bit and I think that in this usecase it would be better to have a minimal image for the box.

That this is not so simple I have already thought.
So you have no interest in it ? Possibly as a team effort?

I alone will not manage this, but with help and some instructions I'm sure to come forward there

Jalle19, 2021-12-17:
There isn't really a team behind this I'm afraid. I maintain a fork which keeps things up to date, but that's about it.

If boot times are an issue then maybe simply don't reboot? I'm not sure a custom image will boot much quicker - it's not like Engima is bloated.

SaulGoodman1337, 2021-12-17:
Yes you are right. The effort obviously just seems too high.
No I actually plan to wake up the box with wake on lan when it is needed. So the whole enigma2 stuff is a thorn in my side and I would like to build an image that throws out everything that is unnecessary for a headless operation. Do you have some experience in this area and know how the buildsystem of for example openatv can be adapted ?

Jalle19, 2021-12-17:
Honestly no, I've never tried building Enigma myself, much less modifying it.

BigDi, 2021-12-20:
I used OpenATV which comes with minisapip. You can also run it headless.

walterav1984, 2021-12-23:
For a server/cmd only image without enigma2 you might try github.com/OpenVisionE2

https://openvision.tech/?p=203

@SaulGoodman1337 It seems the xtrend/et1x000 is listed.

SaulGoodman1337, 2021-12-25:
> I used OpenATV which comes with minisapip. You can also run it headless.

As i Said, this is exactly what im doing ...

> For a server/cmd only image without enigma2 you might try github.com/OpenVisionE2
> 
> https://openvision.tech/?p=203
> 
> @SaulGoodman1337 It seems the xtrend/et1x000 is listed.

Brilliant!! This is what i'm searching for! Thanks i will try

walterav1984, 2022-04-30:
Although a while ago the CMD images went live actually 2 weeks ago, if you are still interested:
https://images.openvision.dedyn.io/11.2/CMD/Xtrend/et10000/

There is a thread on the [openvisiontech forum](https://forum.openvision.tech/viewtopic.php?t=1116) about CMD images in the request section. You may even request a new `minisatip` package for your build like 1.20b which has alot of fixes for enigma2 devices.  

One real note is that the CMD image miss a `opkg` feed location, however you can copy paste the `/etc/opkg/conffiles` from the VISION image to get `opkg` running.

lars18th, 2023-05-18:
Hi @walterav1984 ,

The forum at https://forum.openvision.tech/ is closed? I can't register! 🤔