perexg/satip-axe #192: Digibit R1 network doesn't work with satip-axe

seebaerhh, 2022-04-12:
What I did:
-  activated DHCP Client in the original firmware
-  copied satip-axe-201810211549-15.usb on a 4GB USB stick and renamed it as 'idl4k.scr' 
-  copied satip-axe-201810211549-15.fw (and didn't rename it)
-  rebooted my Digit R1 -> knight rider led mode -> the network led starts to permanently glow (running under stock firmware it flashes multiple times until it stays on)
- can't ping the Digit R1, Asus router shows 'eth2 Link UP, Link Down, Link UP, Link Down' - but it obviously doesn't acquire an IP address. Doesn't show up under connected devices.

Is there any solution to this problem? Seems like the ethernet interface doesn't properly work under satis-axe firmware!?

Jalle19, 2022-04-12:
So the link light on your router doesn't stay on?

seebaerhh, 2022-04-12:
I didn't monitor the LED on the router. The LAN-LED on the Digibit R1 is permanently on but the device doesn't get an IP address from the router. Everything works flawlessly when running the stock firmware.

Jalle19, 2022-04-13:
Are you sure you're pinging the right IP address? It might not get the same IP as it did with the stock firmware. Try flashing again too if it helps.

Jalle19, 2022-04-19:
@seebaerhh did you get it to work?

klada, 2022-04-19:
@seebaerhh Did you purchase the Digibit R1 as a used device or has it ever been used with the custom satip-axe firmware before?

My guess is that the network config could be set to "static" in the satip-axe config partition. In this case you can try either one of the two steps:

- On your local machine, use the "arp" command to temporarily assign a suitable IP address to the Digibit in the ARP cache of your local machine: (arp -s 192.168.1.2 -i ethX aa:bb:cc:11:22:22, then try to ping the Digibit)

- Or you could also try to move idl4k.cfgresetusb as idl4k.scr to your USB stick to reset any previous configuration (if any). See the "Recovery from the wrong configuration" entry in the README file.

seebaerhh, 2022-04-19:
Hi @klada and @Jalle19!
Thank you very much for your comments.
For now I'm running the Digibit R1 with the original firmware. 
I will soon try again with Satie-axe and take your advice into account. Unfortunately I won't be able to do that for a few days. I will keep you updated here.

Jalle19, 2022-04-19:
Good point @klada!

mas0nwest, 2022-07-18:
Did this get resolved?

seebaerhh, 2022-07-18:
> 

No, unfortunately. I sold the Digibit.

perexg, 2023-01-26:
Seems stale.

closed 2023-01-26 by perexg