perexg/satip-axe #197: NO NETWORK

VenimK, 2023-09-25:
DIGIBIT R1

WHEN booting stock firmware no network (and device gets in a loop)
WHEN booting with usb and idl4k.recovery device boots stock firmware OK

When booting with USB and using satin.exe NO NETWORK

What's going ON ???

Jalle19, 2023-09-25:
Sounds like something is wrong with your network, or maybe you're expecting it to get a specific IP address but it's getting some other address?

VenimK, 2023-09-26:
All ready try 'd static IP and DHCP ON stock firmware as well ...

Jalle19, 2023-09-26:
Well, if you can't get the stock firmware to work then I don't think you'll get any custom firmware working either.

VenimK, 2023-09-26:
> Well, if you can't get the stock firmware to work then I don't think you'll get any custom firmware working either.

So device broken then :(

Jalle19, 2023-09-26:
You can try my builds instead from here: https://github.com/Jalle19/satip-axe/releases/tag/build25, although I have a feeling it won't help.

jonathanmbradshaw, 2023-09-26:
Interesting,  I have the same hardware and I have started to see a lot of errors at the ethernet interface on the LAN hub.  if you have access to the counters at this level it may be worth a look.

VenimK, 2023-09-26:
> You can try my builds instead from here: https://github.com/Jalle19/satip-axe/releases/tag/build25, although I have a feeling it won't help.

This morning booted with with recovery on USB
Then installed your latest FW directly in de webmin
And it booted ok
For now !!!

Jalle19, 2023-09-26:
Glad to hear!

VenimK, 2023-09-29:
Has Been working OK for 3 Days
Now it does the same thing ???
I guess it 's really broke
what would be a good replacement ????

palik42, 2024-03-10:
Hi, i have same issue. i using some older version of satipaxe..  I've found 2 workarounds.
1. connect your digibit r1 directly to router( switch have issue, i've tried zyxel (not working with 1000mb port, and netgear - sometime is ip loading long time sometime need restart)  with direct connection to router were these issues minimized for me.
2. you can set port on switch/router to 100mbit and try it.

palik42, 2024-03-10:
for me it looks like ip setup is fighting between stock fw and satipaxe. i dont remember my setup because i was playing it but you can try 3 scenarios: 
1.  original dhcp+satipaxe static e.g. x.x.x.86 ..
2. original x.x.x.86 +satipaxe x.x.x.86
3. original x.x.x.85 +satipaxe x.x.x.86

sometime i have issue to but it is somehow working.

palik42, 2024-03-12:
I've just updated to original dhcp+satipaxe dhcp with [Jalle19]build25
same behavior as original dhcp+satipaxe static e.g. x.x.x.86 after some time is network working.

You can also try diferent ethernet cable - this could be issue too - one my cable wasn't able to start network.

But still is best way to connect digibit r1 directly to my router.

nrpetonr, 2025-05-31:
similar issue. i believe i am able to reflash the image succesfuly, however its network does not seem to work. i am able to collect logs using uart, unfortunately i am not able to find the solution, despite using any possible combination of flashing only with usb, to flash, resetting config or using recovery file. see logs hereianfter. any idea?

```
idl4k> printenv
bootdelay=1
baudrate=115200
board=idl4k
monitor_file_name=u-boot.bin
load_addr=0x84000000
autostart=yes
silent=1
resetenv=eeprom write $load_addr 0x00080000 1000
updatefw=usb start;set filesize;fatload usb 0 $load_addr idl4k.scr;if test "${filesize}" != ""; then autoscr $load_addr; fi 
mtdids=nand0=gen_nand.1
mtdparts=mtdparts=gen_nand.1:128k(env-nand0),4M(system-nand0),32M(fw1-nand0),32M(fw2-nand0),-(data-nand0)
bootargs=console=ttyAS0,115200 bigphysarea=20000 axecfg=reset
filesize=D9CE9F
partition=nand0,0
mtddevnum=0
mtddevname=env-nand0
bootcmd=run updatefw;nboot.jffs2 $load_addr fw2-nand0
satip-axe=1
ethaddr=00:24:A6:00:2B:A9
stdin=serial
stdout=serial
stderr=serial
```

```
idl4k> dhcp
Using MAC Address 00:24:A6:00:2B:A9
STM-GMAC: Unable to find a known PHY!
STM-GMAC: info: PHY at address=0x00 has ID=0x001cc915
STM-GMAC: info: PHY at address=0x01 has ID=0x001cc915
STM-GMAC: ERROR: no PHY detected
```

VenimK, 2025-06-01:
i left it alone for a few days.
started again, afterwards ???

nrpetonr, 2025-06-01:
> i left it alone for a few days. started again, afterwards ???

let's see, i am not that confident. did you also have this error?

i```
dl4k> dhcp
Using MAC Address 00:24:A6:00:2B:A9
STM-GMAC: Unable to find a known PHY!
STM-GMAC: info: PHY at address=0x00 has ID=0x001cc915
STM-GMAC: info: PHY at address=0x01 has ID=0x001cc915
STM-GMAC: ERROR: no PHY detected
```

palik42, 2025-06-01:
@perexg @Jalle19 Please could you check this if it is possible to fix it?
Thank you very much.

Jalle19, 2025-06-01:
I'm unable to reproduce these issues so I don't think I can help much.

nrpetonr, 2025-06-01:
i tried to reflash 'fw1-nand0' partition with original firmware using below commands, it succeeded but result is the same reagrding the network, see log.  i am afraid it is hardware issue, unless i am just overlooking some some stupid mistake in my recovery procedures.

set filesize
fatload usb 0 $load_addr idl4k.bin
nand erase fw1-nand0
nand write.jffs2 $load_addr fw1-nand0 $filesize

```
idl4k> dhcp
Using MAC Address 00:24:A6:00:2B:A9
STM-GMAC: Unable to find a known PHY!
STM-GMAC: info: PHY at address=0x00 has ID=0x001cc915
STM-GMAC: info: PHY at address=0x01 has ID=0x001cc915
STM-GMAC: ERROR: no PHY detected
```

VenimK, 2025-06-01:
Remove the router
Or remove the ip from digivit in the router

nrpetonr, 2025-06-01:
did so, tried w/o any router, tried w/ different router, tried w/ switch, still the behaviour is the same.  chatGPT suggestion :-)

Resolution Steps
1. Update U-Boot PHY Driver
Patch U-Boot: Apply commits that fix RTL8211E recognition:

text
static struct phy_driver RTL8211E_driver = {
    .name = "RealTek RTL8211E",
-   .uid = 0x1cc[9](https://patchwork.ozlabs.org/project/uboot/patch/1458588547-2555-1-git-send-email-merker@debian.org/)[10](https://patchwork.ozlabs.org/patch/449385/),
+   .uid = 0x1cc915,
-   .mask = 0xffffff,
+   .mask = 0xfffff0,
    ...
};
Rebuild U-Boot and redeploy to the device.

2. Modify Device Tree
Ensure the PHY node specifies the correct address, interrupts, and compatible string:

text
&mdio {
    ethphy0: ethernet-phy@0 {
        compatible = "realtek,rtl8211e";
        reg = <0>;
        interrupts = <25 IRQ_TYPE_LEVEL_LOW>;
        interrupt-parent = <&gpio0>;
        reset-gpios = <&gpio0 12 GPIO_ACTIVE_LOW>;
        reset-assert-us = <10000>;
    };
};
3. Manual PHY Initialization
If automatic detection fails, manually initialize the PHY in U-Boot:

bash
mii device GMAC
mii info 0x00  # Verify PHY at address 0x00
mii modify 0x00 0x1f 0x0000  # Reset PHY
mii modify 0x00 0x00 0x8000  # Software reset
4. DHCP Workaround
Temporarily use a static IP if DHCP remains unstable after PHY recognition:

bash
setenv ipaddr 192.168.1.100
setenv serverip 192.168.1.1
saveenv
Verification
Check PHY registration in U-Boot logs:

text
Net:   eth0: STM-GMAC, eth1: STM-GMAC
PHY: 0:00 - Link is Up - 1000/Full
Test network connectivity:

bash
ping 192.168.1.1
Conclusion
The error arises from U-Boot’s inability to map the detected PHY ID (0x001cc915) to the RTL8211E driver due to masking or configuration issues. Updating the driver’s PHY ID/mask, correcting the device tree, and ensuring proper reset sequences resolve the problem. For custom firmware builds (e.g., satip-axe), ensure these fixes are backported to the embedded U-Boot version.

nrpetonr, 2025-06-01:
guys, solved by erasing  data-nan0. thanks to all for support, appreciate it.